cturvey/stm32extldr #4: CLIVEONE-W25Q256_STM32H7XX-PB2-PG6-PF8-PF9-PF7-PF6.flm

muaxiaohei, 2023-10-17:
hello!!!
Can you please add a CLIVEONE-W25Q256_STM32H7XX-PB2-PG6-PF8-PF9-PF7-PF6.flm configuration?
Also, i have a question, is there a plan to add a dual QSPI download algorithm?

cturvey, 2023-10-17:
Added
 https://github.com/cturvey/stm32extldr/blob/main/h7_w25q256/W25Q256_STM32H7XX-CUSTOM17.FLM

Make the dual request via PayPal Contribution with a pin list

sourcer32@gmail.com
  *  https://paypal.me/cliveone (Preferred)
  *  https://www.venmo.com/u/cliveone

cturvey, 2023-10-19:
Did the new version work for you?

cturvey, 2023-10-20:
Did it fail with an error message? What did that say?
Did it write the wrong content? Please elaborate so I might troubleshoot this.
Is there an available UART for debug/diagnostic output? Which UART and pins?

muaxiaohei, 2023-10-20:
Stop in erase stage.
[image]

cturvey, 2023-10-21:
Ran a version here with alternate pins, and RAM for Algorithm 0x20000000/0x0FF00, erased and wrote 9MB

muaxiaohei, 2023-10-21:
Do you mean to configure it like this? Cannot work too...🥲
[image]

cturvey, 2023-10-21:
Yes, confirming pins
CLK:PB2:AF9
NCS:PG6:AF10
D0:PF8:AF10
D1:PF9;AF10
D2:PF7:AF9
D3:PF6:AF9
What are the pins for the dual configuration? Do you have a schematic you can share?
These are authentic Winbond W25Q256 parts? What JEDEC ID is reported?

muaxiaohei, 2023-10-21:
Thanks for your patience！😆
i will confirm the pins and upload the schematic tomorrow!

muaxiaohei, 2023-10-22:
Here is a screenshot. If you want, I can upload a schematic. The MCU is STM32H750XBH6.
Besides, I only use U7 currently.
[image]
JEDEC ID read out as below:
[image]