cturvey/stm32extldr #5: External Loader for G0 and G4 for W25Qxx

Guilherme-Mello, 2024-01-28:
Hello, i'm nedded of the external loaders for some of the MCUs from the G0 and G4 family.

The most "urgent" and specific REFs for me is:

- **STM32G431CBU6*** and **STM32G431KBT6***
- STM32G030K6T6 and STM32G031K6T6
- STM32G071CBT6 and **STM32G071RBT6***

All of them in winbond's:

**- W25Q32*** 
- W25Q64
**- W25Q128***
- W25Q256

**The refs in * is the even most urgent for me. If you can help me with this i would be very greatful!**

cturvey, 2024-01-28:
Define the pin connectivity for the assorted platforms. 
All the Winbond devices share a common core at this point and are selectively compiled so they report correct name and sizing to STM32 Cube Programmer.

Guilherme-Mello, 2024-01-28:
For right now, i only have the pinout for the STM32G071RBT6 (I'm working on the other plataforms right now) basically the STM32G071RBT6 would be the same that uses that morpho display shild, which has the memory pins mapped to the PC2 - MISO, PC3 - MOSI, PB13 - SCK and PB9 - CS.   The memory chip i have now for testing is a W25q32, but the PCB prototype will have the 128Mb size one.       Hope u can help me :)

Guilherme-Mello, 2024-01-28:
And basically the aplication is for using with displays as a "image memory" for the GUI data.

cturvey, 2024-01-28:
The code base currently is aimed as QUADSPI and OCTOSPI attached devices as these map into the device's primary memory map.
SPI is possible, just with less immediacy.

Guilherme-Mello, 2024-01-28:
For the STM32G070KBTx and/or the STM32G0B0KBTx, the pinout is MISO - PA9, MOSI - PA10, SCK - PA0 and CS - PC6

Guilherme-Mello, 2024-01-28:
> The code base currently is aimed as QUADSPI and OCTOSPI attached devices as these map into the device's primary memory map. SPI is possible, just with less immediacy.

So the communication would be slower, only that, right?

Guilherme-Mello, 2024-01-28:
For the STM32G431CBUx the pinout would be PB15 - MOSI, PB14 - MISO, PB13 - SCK and PB12 - CS (Do you know if the SPI for the G431 is great for flash memory?)

Guilherme-Mello, 2024-01-28:
For the STM32G431CBUx the pinout would be PA7 - MOSI, PA6 - MISO, PA5 - SCK and PC4 - CS (Do you know if the SPI-1 for the G431 is great for flash memory?)   Resent this because i wrote the pinout wrong last time.

Guilherme-Mello, 2024-01-30:
> > > SPI is possible, just with less immediacy.
> > > Means I haven't had cause to port SPI usage, all my targets use the QUADSPI, or OCTOSPI peripheral.
> > > Do you know if the SPI-1 for the G431 is great for flash memory?
> > > It will be like other SPI use cases, as it is not memory mapped you'll have to stage content in RAM prior to using for graphics, data, code, etc.

Ah, ok :)