cturvey/stm32extldr #6: Modify h7_s25sl256/CLIVEONE-S25FL256S_STM32H7XX-PB2-PB6-PD11-PD12-PE2-PD13.stldr to use different chip select

florianhumblot, 2024-04-19:
Hi, 
I'd like to use the flashloader that you have made, it seems like it matches my flash and my MCU, except that we have a slightly different pin configuration.

We have the following:

```
    PE2     ------> OCTOSPIM_P1_IO2
    PB2     ------> OCTOSPIM_P1_CLK
    PB10     ------> OCTOSPIM_P1_NCS
    PD11     ------> OCTOSPIM_P1_IO0
    PD12     ------> OCTOSPIM_P1_IO1
    PD13     ------> OCTOSPIM_P1_IO3
```

As far as I can tell, the only difference is that you're using NCS on PB6, with alternate function 10, while we use NCS on PB10, with alternate function 9.
Would you be able to provide an alternative build with the above chip-select? :)

cturvey, 2024-04-19:
The H7XX version is for QSPI models, H743,H745,H747,H750,H753,H755
OSPI H72x, H73x  or H7Ax. H7Bx
Which STM32 are you using?
Not sure I've got an OCTOSPI build for the Infineon, but can be done
-Clive

florianhumblot, 2024-04-19:
Ah I see, I'm using the STM32H735VGT6 with OCTOSPI indeed

cturvey, 2024-04-20:
PB2:AF9       OCTOSPIM_P1_CLK
  PB10:AF9      OCTOSPIM_P1_NCS
  PD11:AF9      OCTOSPIM_P1_IO0
  PD12:AF9      OCTOSPIM_P1_IO1
  PE2:AF9       OCTOSPIM_P1_IO2
  PD13:AF9      OCTOSPIM_P1_IO3

https://github.com/cturvey/stm32extldr/blob/main/h7_s25sl256/CLIVEONE-S25FL256S_STM32H72X-PB2-PB10-PD11-PD12-PE2-PD13.stldr

florianhumblot, 2024-04-22:
In the end this version seems to be the one we need: https://github.com/cturvey/stm32extldr/blob/main/h7_s25sl256/CLIVEONE-S25FL256S-HYB_STM32H72X-PB2-PB10-PD11-PD12-PE2-PD13.stldr
Thanks!

closed 2024-04-22 by florianhumblot