cturvey/stm32extldr #7: Request: h7_w25q128 PB2,PB10 PD11,PD12,PE2,PD13

CSharperMantle, 2024-08-14:
First of all, thank you for this effort to provide all these loaders.

It would be of great convenience if a configuration of this following could be provided for H7+W25Q128:

* CLK: PB2
* nCS: PB10
* IO0: PD11
* IO1: PD12
* IO2: PE2
* IO3: PD13

Many thanks!

cturvey, 2024-08-14:
Try this, should be suitable for the QSPI models, ie H74x / H75x parts
https://github.com/cturvey/stm32extldr/blob/main/h7_w25q128/CLIVEONE-W25Q128_STM32H7XX-PB2-PB10-PD11-PD12-PE2-PD13.stldr
For H7Ax / H7Bx and H72x / H73x alternate versions will need to be built
-Clive

CSharperMantle, 2024-08-16:
> Try this, should be suitable for the QSPI models, ie H74x / H75x parts
> https://github.com/cturvey/stm32extldr/blob/main/h7_w25q128/CLIVEONE-W25Q128_STM32H7XX-PB2-PB10-PD11-PD12-PE2-PD13.stldr

Yeah I think this would do the job. Many 👍s for your swift response!

closed 2024-08-16 by CSharperMantle