cturvey/stm32extldr #8: Issues with Loader in CubeProgrammer

Austina1or, 2024-08-20:
Hello,
I'm using the board Nucleo-H755ZI and connected an MX25L51245G Flash.
In a test application the QSPI-interface worked fine in dummy mode and my pinning is the following:
CLK:    PF10
CS:      PG6
IO_0:   PF8
IO_1:   PF9
IO_2:   PE2
IO_3:   PF6

Therefore I was expecting that your version "CLIVEONE-MX25L512_STM32H72X-PF10-PG6-PF8-PF9-PE2-PF6" shall work with this setup, am i correct?
Though when I connect in CubeProgrammer with this external loader, I can't ready any data at 0x09000000 or perform any other action with the flash. Also when I probe the clock pin with an oscilloscope, there is no action.
Am I doing anything completely wrong or do I miss a step?
Thanks a lot!

cturvey, 2024-08-20:
Well the H72x parts are OCTO SPI whereas the H75x are QUAD SPI, the MCU is different,  you need a different build.
I'll check later when at a computer.

Austina1or, 2024-08-20:
Ah okay, i was a bit confused by the README content:
"Support for Macronix MX25L512 compatible parts on STM32H723 / H72X / H7Ax / H7Bx platforms, QUAD device on OCTO and STM32H74x / H75x platforms QUAD"
which mentions H75x paltforms QUAD with the matching FLASH device.

Thanks in advance!

cturvey, 2024-08-20:
Well the pin selection is odd for the NUCLEO-H755ZI, as it typically exports on the header via
```
PB2:AF9 CLK
PG6:AF10 NCS
PD11:AF9 D0
PD12:AF9 D1
PE2:AF9 D2
PD13:AF9 D3
```
For
```
PF10:AF9 CLK
PG6:AF10 NCS
PF8:AF10 D0
PF9:AF10 D1
PE2:AF9 D2
PF6:AF9 D3
```
Try
https://github.com/cturvey/stm32extldr/blob/main/h7_mx25l51245g/CLIVEONE-MX25L512_STM32H7XX-PF10-PG6-PF8-PF9-PE2-PF6.stldr

Austina1or, 2024-08-21:
Thanks for your effort!
Unfortunately there's still no action on the Clock pins with the new .stldr-File, i also checked the other possible Clock pins during the "Read attempt" at 0x09000000.
Are there any requirements on the user application already running on the STM?
Or shall the External Loader e.g. even work on an "empty" STM-chip just by connecting via CubeProgrammer with the respective External Loader?
I understood the whole process in a way that by connecting to the board with the external loader, the loader is loaded and run on the RAM automatically with no additional steps needed, am I correct?
I'm a bit lost because I don't know where to tackle this problem.

[image]

cturvey, 2024-08-21:
The memory is at 0x90000000 not 0x09000000

closed 2024-08-21 by Austina1or

cturvey, 2024-09-05:
Please consider contributing to the project for time/materials..

  *  https://paypal.me/cliveone (Preferred)
  *  https://www.venmo.com/u/cliveone
  *  https://buymeacoffee.com/cliveone