cturvey/stm32extldr #10: Request for .stldr File Generation for STM32H745ZI-Q with W25Q128JV Flash Memory

keremzorer, 2024-09-18:
Hi,
I am trying to generate a .stldr file for the STM32H745ZI-Q device, paired with the external flash memory model W25Q128JV. Below is the pin configuration I am using for QUADSPI:

- PB2 → QUADSPI_CLK
- PC11 → QUADSPI_NCS
- PE7 → QUADSPI_IO0
- PE8 → QUADSPI_IO1
- PE9 → QUADSPI_IO2
- PE10 → QUADSPI_IO3

I am encountering issues during this process and would appreciate guidance on what might be going wrong. Additionally, if possible, could you provide a .stldr file for the above configuration?

Thank you for your support!

cturvey, 2024-09-18:
Yeah, will take a gander this evening.

PB2:AF9 CLK
PC11:AF9 BK2 NCS
PE7:AF10 BK2 IO0
PE8:AF10 BK2 IO1
PE9:AF10 BK2 IO2
PE10:AF10 BK2 IO3

My hot-take would be an issue with the pin settings, or the second bank select

cturvey, 2024-09-19:
Try this
https://github.com/cturvey/stm32extldr/blob/main/h7_w25q128/CLIVEONE-W25Q128_STM32H7XX-PB2-PC11-PE7-PE8-PE9-PE10.stldr

kzorer, 2024-09-19:
I uploaded directly into "..\STM32Cube\STM32CubeProgrammer\bin\ExternalLoader" file but it does not work. It gives an error such that: "Error: Data read failed"

cturvey, 2024-09-19:
Ok, will check over later
Has this design been tested/validated with your own BSP code? Bringing up the QSPI on the application side?

cturvey, 2024-09-19:
Perhaps can you screen shot this in STM32 Cube Programmer, with Verbose Level 3, to provide additional context.

kzorer, 2024-09-19:
Yes I have already been tested QSPI peripheral. I'm reading and writing to the flash properly. Here is the screen shot from STM32CubeProgrammer

[image]

cturvey, 2024-09-19:
Ok, so failing in Init()
What HSE source / frequency?
SMPS power supply?

kzorer, 2024-09-20:
Hi again,
My HSE source frequency is 8MHz.
SMPS power supply --> 12V/1A
JEDEC ID --> 7018h that means W25Q128JV-IM*/JM* model reported.

cturvey, 2024-09-20:
Ok, updated loader, was looking for EF 60/40 xx patterns, not EF 70 xx
Please try the new build
Do you have USART available for diagnostics? Which USART and Pins?

cturvey, 2024-09-20:
I have a relatively flat approach using the HAL libraries to create a "DLL" exporting the loader functions and without using interrupts.
Use make + GNU/GCC to compile, and process the STLDR file
No startup.s or main.c, so no Vectors, Reset_Handler, no main()
Loader_Src.c / Loader_Src.h containing the loaders entry points, QSPI code, pin definitions and part details
Dev_Inf.c / Dev_Inf.h containing the headering details
ExternalLoader.ld for the linker template, see examples in next level up.
I can optionally use a UART to output debug/telemetry to understand issues and flow dynamics