cturvey/stm32extldr #12: Interested in a port of this loader: PB2,PC11,PD11,PD12,PA7,PD13 (H7Ax, H7Bx QUAD on OCTO) for W25Q128

ssebikes, 2025-02-15:
Hello,

Every ST community forum post about external flash loaders has brought me here. My configuration is close to an existing .stldr but doesn't quite match. I was wondering if it's possible to get a new version

STM32H7A3 (H7Ax, H7Bx QUAD on OCTO) & W25Q128

    PB2   CLK
    PC11  NCS
    PD11  IO0
    PD12  IO1
    PA7   IO2
    PD13  IO3

Thanks for the work you've put into this. Happy to contribute to the coffee or 3 fund

cturvey, 2025-02-15:
Try
https://github.com/cturvey/stm32extldr/blob/main/h7_w25q128/CLIVEONE-W25Q128_STM32H7AX-PB2-PC11-PD11-PD12-PA7-PD13.stldr

ssebikes, 2025-02-18:
Thanks a lot! I haven't been able to read from the SPI flash yet. This whole process is a huge learning curve, do these steps appear correct?

1. Set up project in STM32CubeIde and assign pins in .ioc for octospi quad-spi.
2. Replace the generated .ld linker file with the [ExternalLoader.ld](https://github.com/cturvey/stm32extldr/blob/main/ExternalLoader.ld) from this repo.
3. Generate bin in post build step.
4. In STM32CubeProgrammer connect to the MCU and include the given .stldr file.
5. Select the generated binary file, program the device, and read the device memory at 0x90000000.

I don't have any of the Loader_Src.c, Dev_Inf.h, Dev_Inf.c files, nor do I have an Init(void) function. My assumption was these are executed from the .stldr driver but I am not sure.