cturvey/stm32extldr #15: Need the loader with STM32H750VBT6 + W25Q64JV

bhushan4520, 2025-04-13:
Hi,
Thanks in advance for your huge efforts. 

I tried but  failed multiple times to make the stldr or FLM file for the following combination.

STM32H750VBT6 + W25QQ64JV

Pin assignment as follows 

PB2    -> QUADSPI_CLK
PB10  -> QUADSPI_BK1_NCS
PC9   ->  QUADSPI_BK1_IO0
PD12 ->  QUADSPI_BK1_IO1
PE2   ->  QUADSPI_BK1_IO2
PD13 -> QUADSPI_BK1_IO3

I apprecaite if you can help me out with both .FLM as well as .stldr because I  am using KEIL and cubeIDE both for differnt projects.

cturvey, 2025-04-13:
Try
https://github.com/cturvey/stm32extldr/blob/main/h7_w25q64/CLIVEONE-W25Q64_STM32H7XX-PB2-PB10-PC9-PD12-PE2-PD13.stldr

bhushan4520, 2025-04-13:
[image]

cturvey, 2025-04-13:
Not sure the build process has changed here. Not tried v2.19
The first one you refer to the W25Q128 not the W25Q64
For the .FLM try
https://github.com/cturvey/stm32extldr/blob/main/h7_w25q64/W25Q64_STM32H7XX-CUSTOM54.FLM
Be sure to expand the RAM for the flash algorithm

bhushan4520, 2025-04-14:
Thanks for your support.. .stldr file worked fine with Cubeprogrammer with version 2.18. 

Now will check with Keil debuggerwith  .FLM

closed 2025-04-14 by bhushan4520