cturvey/stm32extldr #16: Stm32u5G9 with W25Q32 flash

robin171, 2025-07-17:
Hi Clive,

I found this repository through the ST community. 

Do you have time to make a flashloader for me for the STM32U5G9? I have the following hardware configuration:

- W25Q32 flash connected to octospi P0 - PC3, clk to PA3 and CS to PC11
- 3V3_EN to PB1 which needs to be high before flash is powered

Of course I'll make a donation :-)

Kind regards, 

Robin van den Berg

cturvey, 2025-07-18:
Hi Robin,

I'll take a look, can you confirm the data pins"P0 - PC3" is a bit confusing, are you using the upper pair? Can you attach your initialization code for interface and pins? Can you Read the memory, and READ ID, in your current BSP/CSP code?

PA3:AF10 OCTOSPIM_P1_CLK
PC11:AF5 OCTOSPIM_P1_NCS

PC1:AF10 OCTOSPIM_IO4
PC2:AF10 OCTOSPIM_IO5
PC3:AF10 OCTOSPIM_IO6
PC4:AF10 OCTOSPIM_IO7

Thanks,
-Clive

cturvey, 2025-09-13:
Hi Robin,

The Init() can fail on the second call if the configuration of the QSPI device mode is incompatible with the mode used to probe it.
You'd have to check the exit paths, Perhaps instrument thru a UART, or don't change the operational mode. The part itself won't report an error, it just won't sequence/respond as expected.

Pretty sure the Q32 won't support the Status Register 2 write command, but rather as a pair.
Also not large enough to use 4-byte addressing.

Here is a tentative build for the U5F9/U5G9 platform
https://github.com/cturvey/stm32extldr/blob/main/u5_w25q32/CLIVEONE-W25Q32_STM32U5F9-PA3-PC11-PC1-PC2-PC3-PC0.stldr

robin171, 2025-09-15:
Hi Clive, 

Thanks. I will test it tomorrow when I am back in the office

Cheers,

Robin